arduino-libraries/Arduino_BMI270_BMM150 #25: how to increase the sample rate

lixinghe1999, 2023-08-08:
Dear everyone, 

I am struggling in increasing the sample rate of BMI270 on Arduino nano BLE sense Rev2.

I slightly modify the example of examples/interrupts, and I change the line of ODR.
It turns out my code can successfully go to 200Hz, but fail to go to 400Hz or higher.
It is hard to believe it occurs because of the upper bound of Arduino, since I can achieve much better sample rate on others.
I attach my code as follows:
```
#include "Arduino_BMI270_BMM150.h"

class MyBoschSensor: public BoschSensorClass {

  public:
    MyBoschSensor(TwoWire& wire = Wire) : BoschSensorClass(wire) {};

  protected:
    virtual int8_t configure_sensor(struct bmi2_dev *dev)
    {
      int8_t rslt;
      uint8_t sens_list[2] = { BMI2_ACCEL, BMI2_GYRO };

      struct bmi2_int_pin_config int_pin_cfg;
      int_pin_cfg.pin_type = BMI2_INT1;
      int_pin_cfg.int_latch = BMI2_INT_NON_LATCH;
      int_pin_cfg.pin_cfg[0].lvl = BMI2_INT_ACTIVE_HIGH;
      int_pin_cfg.pin_cfg[0].od = BMI2_INT_PUSH_PULL;
      int_pin_cfg.pin_cfg[0].output_en = BMI2_INT_OUTPUT_ENABLE;
      int_pin_cfg.pin_cfg[0].input_en = BMI2_INT_INPUT_DISABLE;

      struct bmi2_sens_config sens_cfg[2];
      sens_cfg[0].type = BMI2_ACCEL;
      sens_cfg[0].cfg.acc.bwp = BMI2_ACC_OSR2_AVG2;
      sens_cfg[0].cfg.acc.odr = BMI2_ACC_ODR_400HZ;
      sens_cfg[0].cfg.acc.filter_perf = BMI2_PERF_OPT_MODE;
      sens_cfg[0].cfg.acc.range = BMI2_ACC_RANGE_4G;
      sens_cfg[1].type = BMI2_GYRO;
      sens_cfg[1].cfg.gyr.filter_perf = BMI2_PERF_OPT_MODE;
      sens_cfg[1].cfg.gyr.bwp = BMI2_GYR_OSR2_MODE;
      sens_cfg[1].cfg.gyr.odr = BMI2_GYR_ODR_25HZ;
      sens_cfg[1].cfg.gyr.range = BMI2_GYR_RANGE_2000;
      sens_cfg[1].cfg.gyr.ois_range = BMI2_GYR_OIS_2000;

      rslt = bmi2_set_int_pin_config(&int_pin_cfg, dev);
      if (rslt != BMI2_OK)
        return rslt;

      rslt = bmi2_map_data_int(BMI2_DRDY_INT, BMI2_INT1, dev);
      if (rslt != BMI2_OK)
        return rslt;

      rslt = bmi2_set_sensor_config(sens_cfg, 2, dev);
      if (rslt != BMI2_OK)
        return rslt;

      rslt = bmi2_sensor_enable(sens_list, 2, dev);
      if (rslt != BMI2_OK)
        return rslt;

      return rslt;
    }
};

MyBoschSensor myIMU(Wire1);

float x, y, z;
volatile unsigned int count = 0;
unsigned int time_start;
void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  while (!Serial);
  myIMU.debug(Serial);
  myIMU.onInterrupt(print_data);
  myIMU.begin();

  Serial.print("Accelerometer sample rate = ");
  Serial.println(myIMU.accelerationSampleRate());
  time_start = millis();
}
void print_data() {
  // we can also read accelerometer / gyro data here!
  Serial.println("Got new data!");
  myIMU.readAcceleration(x, y, z);
  count +=1;
}

void loop() {
  if ((millis() - time_start) > 2000)
  {
    Serial.println(count/2);
    count = 0;
    time_start = millis();
  }
  }
```

popoC, 2024-07-03:
I also have the same question, is there any follow-up solution?

popoC, 2024-07-03:
I tried adding _wire->setClock(400000); to BoschSensorClass::begin() at the bottom of bmi270.cpp to speed up i2c, which seems to improve the sampling speed.