arduino-libraries/Arduino_BMI270_BMM150 #37: no member named 'onInterrup' ??

mconsidine, 2024-02-18:
Hi,
I've installed this and the "Simple" examples work.  But the "Interrupts subclassing" generates and error regarding "Wire1" (which I assume should be "Wire"), as well as this 
  Compilation error: 'class MyBoschSensor' has no member named 'onInterrupt'

Can someone advise me how to fix this or what I might be doing wrong?  I'm just trying to get the example to run - no changes.
Thx,
mconsidine

aentinger, 2024-03-07:
This library is intended for the Nano 33 BLE Sense. If you compile for this board via
```bash
arduino-cli compile -b arduino:mbed_nano:nano33ble examples/Interrupts_subclassing -v
```
it compiles without problem.

But regardless of this intentional target there is quite a large range of supplied target architectures, see [here](https://github.com/arduino-libraries/Arduino_BMI270_BMM150/blob/master/.github/workflows/compile-examples.yml#L31-L102). For which board did you compile the example in question?

mconsidine, 2024-03-07:
Thanks for the look.  I think this is a case of operator (ie, my) error ...

Tyenca, 2024-03-12:
Does this mean the code won't work on the arduino uno? Because I am getting the errors regarding Wire1 and no member named  'onInterrupt' too

aentinger, 2024-03-12:
Yes, that's what it means. It would probably pay off to limit the number of supported architectures (in `library.properties`) to those that are actually supported.